ricnaaru/adv_camera #4: Android Open Camera bug

Driaan, 2019-10-29:
I recieve this unhandled exception whenever i open a camera using adv_camera:

W/ActivityThread(26516): handleWindowVisibility: no activity for token android.os.BinderProxy@d044fba
I/flutter (26516): hasPermission => true
W/PlatformViewsController(26516): Creating a virtual display of size: [1080, 1920] may result in problems(https://github.com/flutter/flutter/issues/2897).It is larger than the device screen size: [1080, 1776].
W/Gralloc3(26516): allocator 3.x is not supported
D/tag     (26516): onAttachedToWindow
D/tag     (26516): onResume() called
D/AdvCamera(26516): set Parameters Failed
D/AdvCamera(26516): 1920, 1080
W/System.err(26516): java.lang.RuntimeException: setParameters failed
W/System.err(26516): 	at android.hardware.Camera.native_setParameters(Native Method)
W/System.err(26516): 	at android.hardware.Camera.setParameters(Camera.java:2117)
W/System.err(26516): 	at com.ric.adv_camera.AdvCamera.refreshCameraPreview(AdvCamera.java:536)
W/System.err(26516): 	at com.ric.adv_camera.AdvCamera.refreshCamera(AdvCamera.java:526)

ricnaaru, 2019-10-30:
please note this

> For Android's permission, you have to configure it yourself (using Dexter, etc.). This plugin is made to support my other plugin adv_image_picker, so this permission is handled in adv_image_picker

On Android, you have to request permission manually, you can use dexter or another permission requestor plugin on Java. or you can see my code at [adv_image_picker](https://pub.dev/packages/adv_image_picker#-readme-tab-)

or you can manually set camera permission on Android setting

Driaan, 2019-10-30:
The permission is not the issue, i have already handled that myself. The camera view comes up clearly but on the very first time open camera it is frozen and spits out the above error message and i am unable to take a photo from it. If i switch to front camera it works smoothly and if i switch back to the back camera then it works smoothly again.

ricnaaru, 2019-11-08:
alright, the thing is I found out that android's preview size not all is useable, so maybe i will create one function to find the best preview resolution and set it right away, instead of setting it manually

ricnaaru, 2020-01-06:
Please try the new version

ricnaaru, 2020-12-25:
please try the latest version (1.2.0) and if the problem still persists, please kindly reopen this thread. Thanks!

closed 2020-12-25 by ricnaaru