ricnaaru/adv_camera #6: Unhandled Exception: PlatformException(error, Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference

DesortsSolution, 2019-12-09:
E/flutter (27234): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(error, Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference, null)
E/flutter (27234): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
E/flutter (27234): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:316:33)
E/flutter (27234): <asynchronous suspension>
E/flutter (27234): #2      AdvCameraController.setPreviewRatio (package:adv_camera/controller.dart:97:34)
E/flutter (27234): <asynchronous suspension>

ricnaaru, 2020-01-06:
please note this

> For Android's permission, you have to configure it yourself (using Dexter, etc.). This plugin is made to support my other plugin adv_image_picker, so this permission is handled in adv_image_picker

On Android, you have to request permission manually, you can use dexter or another permission requestor plugin on Java. or you can see my code at [adv_image_picker](https://pub.dev/packages/adv_image_picker#-readme-tab-)

or you can manually set camera permission on Android setting

OliverNarramore, 2020-01-10:
I have added these permissions to the app/src/main/androidmanifest.xml file:

    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-feature android:name="android.hardware.camera.front" />
    <uses-feature android:name="android.hardware.microphone" />
    <uses-feature android:name="android.hardware.camera2" />
    <uses-feature
        android:name="android.hardware.camera.autofocus"
        android:required="true" />
    <uses-feature android:name="android.hardware.camera.flash" />
    <uses-permission android:name="android.permission.FLASHLIGHT" />

and I am still getting the above mentioned error. What am I missing?

closed 2020-03-13 by ricnaaru