ricnaaru/adv_camera #7: Camera crashes android 9 and 10

OliverNarramore, 2020-01-14:
I'm attempting to use this package, but it seems to crash as soon as i take an image on android 9 and 10, but works fine on android 8. This is the error I get:

Unhandled Exception: PlatformException(error, takePicture failed, null)
    #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
    #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:321:33)
    <asynchronous suspension>
    #2      AdvCameraController.captureImage (package:adv_camera/controller.dart:111:19)
    #3      _AdvancedCameraState.build.<anonymous closure> (package:testapp/ui/views/Add%20Image/advanced_camera.dart:265:36)
    #4      _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:706:14)
    #5      _InkResponseState.build.<anonymous closure> (package:flutter/src/material/ink_well.dart:789:36)
    #6      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:182:24)
    #7      TapGestureRecognizer.handleTapUp (package:flutter/src/gestures/tap.dart:486:11)
    #8      BaseTapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:264:5)
    #9      BaseTapGestureRecognizer.handlePrimaryPointer (package:flutter/src/gestures/tap.dart:199:7)
    #10     PrimaryPointerGestureRecognizer.handleEvent (package:flutter/src/gestures/recognizer.dart:467:9)
    #11     PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:76:12)
    #12     PointerRouter._dispatchEventToRoutes.<anonymous closure> (package:flutter/src/gestures/pointer_router.dart:117:9)
    #13     _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:379:8)
    #14     PointerRouter._dispatchEventToRoutes (package:flutter/src/gestures/pointer_router.dart:115:18)
    #15     PointerRouter.route (package:flutter/src/gestures/pointer_router.dart:101:7)
    #16     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:218:19)
    #17     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:198:22)
    #18     GestureBinding._handlePointerEvent (package:flutter/src/gestures/binding.dart:156:7)
    #19     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:102:7)
    #20     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:86:7)
    #21     _rootRunUnary (dart:async/zone.dart:1138:13)
    #22     _CustomZone.runUnary (dart:async/zone.dart:1031:19)
    #23     _CustomZone.runUnaryGuarded (dart:async/zone.dart:933:7)
    #24     _invoke1 (dart:ui/hooks.dart:273:10)
    #25     _dispatchPointerDataPacket (dart:ui/hooks.dart:182:5)

ricnaaru, 2020-01-20:
I will look into it, but this may take a while

ricnaaru, 2020-02-15:
have you checked the permission? is it enabled?

closed 2020-03-13 by ricnaaru