ricnaaru/adv_camera #11: The camera does not stop when the app is minimized

matheustdo, 2020-02-01:
When the app is minimized the camera still running at background, the battery is drained and the mobile flashlight is locked.

ricnaaru, 2020-02-03:
thanks for reporting, i will try to fix it!

ricnaaru, 2020-02-15:
anyway, i checked on mine, it didn't block the flash light and i still can access my native camera apps, can you give me more details, as what version are you using and what device are you testing on? thanks!

mcsugaman, 2020-03-06:
I also have problem with minimization of app...
When I minimize app, where camera is initialized and then return back to  app, camera preview is just frozen. I hear "blink" sound which indicates, that camera has opened, but the preview is frozen and contains picture, that was on preview after I minimized the app. 
Thanks for help.

ricnaaru, 2020-06-03:
please provide the device model, and is it on android / ios?

matheustdo, 2020-06-05:
> anyway, i checked on mine, it didn't block the flash light and i still can access my native camera apps, can you give me more details, as what version are you using and what device are you testing on? thanks!

> anyway, i checked on mine, it didn't block the flash light and i still can access my native camera apps, can you give me more details, as what version are you using and what device are you testing on? thanks!

So, I tested it on an Asus Zenfone 3 device running the 8.0.0 Android version. I'm late, but when I did this test I was using the adv_camera at the ^1.0.0+1.

ricnaaru, 2020-12-25:
please try the latest version (1.2.0) and if the problem still persists, please kindly reopen this thread. Thanks!

closed 2020-12-25 by ricnaaru