ricnaaru/adv_camera #12: Empty path onImageCaptured

jl-gogovapps, 2020-02-08:
Thanks for the great work on this library, it was easy to get started with the flash and camera rotation.

One issue I'm having is that when the picture is taken, the path is empty.  Happy to provide more details, I'll start with that I'm testing on an Android 10 Samsung Galaxy S9.

Is this something that sounds familiar?

cosmindidis, 2020-02-12:
Try calling setSavePath('<location of files>') on the controller, it did the trick for me.

ricnaaru, 2020-02-15:
have you enabled the permission manually? because i didn't handle any permission here, so possibly it could be permission issue, but if it's not, please give me a chuck of your scripts so i can reproduce it here.. thanks!

ricnaaru, 2020-12-25:
please try the latest version (1.2.0) and if the problem still persists, please kindly reopen this thread. Thanks!

closed 2020-12-25 by ricnaaru