ricnaaru/adv_camera #16: Error opening Android camera

danielbrito1987, 2020-04-02:
When trying to call AdvCamera the following error is returning me:

E/flutter (20137): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: PlatformException(error, Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference, null)

I already uninstalled and installed the plugin again and the error continues

ricnaaru, 2020-06-02:
when did the error occur? while opening the camera, or while trying to take picture?

danielbrito1987, 2020-06-02:
> when did the error occur? while opening the camera, or while trying to take picture?

while opening the camera

ricnaaru, 2020-06-03:
please try the latest version

closed 2020-06-03 by ricnaaru