ricnaaru/adv_camera #20: Focus

mcsugaman, 2020-05-16:
Hi,

is it possible to force camera focus on camera preview?
Now we have to tap screen for focus.
Thanks.

shachneff, 2020-05-17:
> Hi,
> 
> is it possible to force camera focus on camera preview?
> Now we have to tap screen for focus.
> Thanks.

+1

ricnaaru, 2020-06-02:
what do you mean by force camera focus? so initially the camera focus is locked at center?

shachneff, 2020-06-02:
> what do you mean by force camera focus? so initially the camera focus is locked at center?

if you don’t tap with your finger in the live-preview screen, the photo will be cloudy (not in focus)

mcsugaman, 2020-06-02:
> > what do you mean by force camera focus? so initially the camera focus is locked at center?
> 
> if you don’t tap with your finger in the live-preview screen, the photo will be cloudy (not in focus)

+1

Eswarrajpurohit, 2020-11-03:
hey i have used stack to overlay something over this widget ,so my question is that ,is there any way to add custom focus point,it would be very kind of you helping me out here

ricnaaru, 2020-12-25:
this topic is related with #42, please refer to that thread from now on, i will close this one, thanks!

closed 2020-12-25 by ricnaaru