ricnaaru/adv_camera #23: Camera view is not showing on iPhone 6s plus

andriy-gulak-ew, 2020-06-03:
I've tried to run the example you provided without any changes in code.
The problem is detected on iOS on iPhone 6s plus device. 

As you can see on the screenshot

 below, the camera is not showing a preview, but functionally it works: the camera switch is working and the image capturing is fine. 

The only problem is in the camera preview which is not working without any errors in logs.
Here is the Flutter doctor output:
[✓] Flutter (Channel master, 1.19.0-2.0.pre, on Mac OS X 10.15.2 19C57, locale en-UA) 
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 11.3)
[✓] Chrome - develop for the web
[✓] Android Studio (version 3.6)
[✓] Connected device (4 available)

[image]

ricnaaru, 2020-06-04:
sorry, just have time to reply, another thing you can try is, run it on stable channel of Flutter, that's the latest version that i tried out

JSBmanD, 2020-06-09:
> 
> 
> sorry, just have time to reply, another thing you can try is, run it on stable channel of Flutter, that's the latest version that i tried out

I'm running app on stable channel and preview isn't working either.

ricnaaru, 2020-12-25:
please try the latest version (1.2.0) and if the problem still persists, please kindly reopen this thread. Thanks!

closed 2020-12-25 by ricnaaru