ricnaaru/adv_camera #24: How get the image data without direct save to disk.

PandaKM, 2020-06-03:
hi , richardothayeb. with me the adv_camera is very helpful.
I want to be able to get data directly instead of saving it to disk first, which can better optimize the process. I tried to change your source code, but I failed because of lack of ability.
If you don't mind, can you provide some guidance, or provide a method to get the picture data directly for manual processing. Thanks♪(･ω･)ﾉ

ricnaaru, 2020-06-03:
on android, camera.takePicture(null, null, jpegCallback); the callback will
have byte[] which is bytes array, and on IOS
stillImageOutput.captureStillImageAsynchronously(from:
videoConnection)
closure will have imageDataSampleBuffer which can be used to let imageData
=
AVCaptureStillImageOutput.jpegStillImageNSDataRepresentation(imageDataSampleBuffer)
which
is byte array too

PandaKM, 2020-06-04:
Thank you for your answer. I'm trying.

closed 2020-08-07 by ricnaaru