ricnaaru/adv_camera #25: [ Android ] Multiple camera widgets issue

intellitour, 2020-06-03:
In Android, when you put multiple pages (navigate from one page to another). each with `AdvCamera`, in the second one there will be a blank screen instead of the preview.

I believe this is related to: https://github.com/ricnaaru/adv_camera/issues/22

ricnaaru, 2020-06-04:
Let me check it

intellitour, 2020-06-04:
I managed to catch the error log:

```shell
E/MethodChannel#flutter/platform_views: Failed to handle method call
    android.view.InflateException: Binary XML file line #8: Binary XML file line #8: Error inflating class fragment
    Caused by: android.view.InflateException: Binary XML file line #8: Error inflating class fragment
    Caused by: java.lang.IllegalArgumentException: Binary XML file line #8: Duplicate id 0x7f08004c, tag null, or parent id 0xffffffff with another fragment for com.ric.adv_camera.CameraFragment
        at android.app.FragmentManagerImpl.onCreateView(FragmentManager.java:3621)
        at android.app.FragmentController.onCreateView(FragmentController.java:99)
        at android.app.Activity.onCreateView(Activity.java:6316)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:780)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:730)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:863)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:824)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:515)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:423)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:374)
        at com.ric.adv_camera.AdvCamera.<init>(AdvCamera.java:89)
        at com.ric.adv_camera.AdvCameraFactory.create(AdvCameraFactory.java:20)
        at io.flutter.plugin.platform.SingleViewPresentation.onCreate(SingleViewPresentation.java:179)
        at android.app.Dialog.dispatchOnCreate(Dialog.java:494)
        at android.app.Dialog.show(Dialog.java:342)
        at android.app.Presentation.show(Presentation.java:267)
        at io.flutter.plugin.platform.VirtualDisplayController.<init>(VirtualDisplayController.java:94)
        at io.flutter.plugin.platform.VirtualDisplayController.create(VirtualDisplayController.java:47)
        at io.flutter.plugin.platform.PlatformViewsController$1.createPlatformView(PlatformViewsController.java:112)
        at io.flutter.embedding.engine.systemchannels.PlatformViewsChannel$1.create(PlatformViewsChannel.java:95)
        at io.flutter.embedding.engine.systemchannels.PlatformViewsChannel$1.onMethodCall(PlatformViewsChannel.java:59)
        at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:226)
        at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:85)
        at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:631)
        at android.os.MessageQueue.nativePollOnce(Native Method)
        at android.os.MessageQueue.next(MessageQueue.java:325)
        at android.os.Looper.loop(Looper.java:142)
        at android.app.ActivityThread.main(ActivityThread.java:6942)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)
```

ricnaaru, 2020-06-05:
I see, i'll try to fix the error within this week!

ricnaaru, 2020-07-25:
Sorry this take a long time, i've looked into it, and since it's been quite a long time since I do Android and Fragment, i couldn't get this fixed. It seems like somehow it will become nested Fragment. So, probably I will look into this more, if I have some spare time. Sorry.

closed 2020-08-07 by ricnaaru

intellitour, 2020-08-07:
@ricnaaru by closing you mean `fixed` or `won't fix`?

benneca, 2021-12-10:
This is still an issue, is there a workaround for this? I have this issue, not because of multiple screens with cameras, but rather the first screen (default) is the camera. Due to the dart bug where the initial screen does not call the dispose method, I continually run into the issue above. I have attempted to combat this by trying to use pushreplacenamed on the first item in the stack if the route is the camera screen, this forces the dispose method. However on subsequent page visits, if I want to allow the user to use the back stack, this is not a fully reliable solution. If there were simply a method to check if the camera is already initialized and then resume or dispose of it at that point, this could be resolved.