ricnaaru/adv_camera #26: swicth camera.front

hoverhollydev, 2020-06-06:
Hi Sir I have error in switch camera front.

help me please

I used Android 9  

W/System.err(23964): java.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.List.contains(java.lang.Object)' on a null object reference
W/System.err(23964): 	at com.ric.adv_camera.AdvCamera.translateFlashType(AdvCamera.java:503)
W/System.err(23964): 	at com.ric.adv_camera.AdvCamera.setupCamera(AdvCamera.java:465)
W/System.err(23964): 	at com.ric.adv_camera.AdvCamera.onMethodCall(AdvCamera.java:282)
W/System.err(23964): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:230)
W/System.err(23964): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:85)
W/System.err(23964): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:650)
W/System.err(23964): 	at android.os.MessageQueue.nativePollOnce(Native Method)
W/System.err(23964): 	at android.os.MessageQueue.next(MessageQueue.java:326)
W/System.err(23964): 	at android.os.Looper.loop(Looper.java:160)
W/System.err(23964): 	at android.app.ActivityThread.main(ActivityThread.java:6863)
W/System.err(23964): 	at java.lang.reflect.Method.invoke(Native Method)
W/System.err(23964): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:537)
W/System.err(23964): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

mcsugaman, 2020-06-21:
same problem here,  Samsung Galaxy S9, Android 10 installed

ricnaaru, 2020-06-21:
btw, this shouldn't be happening again, because i remember have fixed it at the latest version, have you tried the latest version?

mcsugaman, 2020-06-21:
yes, I am using version 1.1.1+1. When I call cameraController.switchCamera() I get this error.

mcsugaman, 2020-08-05:
any update on this?

anhtuan23, 2020-08-07:
I have this issue using the current example app too. Tested on Samsung A21s running Android 10.

NguyenQuangKhang, 2020-08-13:
it's Ok in rear but when i switch camera to Front.
in version adv_camera 1.1.2+2, I got this : 
"a virtual display of size: [1631, 2900] may result in problems(flutter/flutter#2897 is larger than the device screen size: [1440, 2900]."

kirya355, 2020-09-11:
Not working, here the code
`I/Camera  (27906): close camera: 0, package name: com.ric.adv_camera_example
I/HwCameraUtil(27906): notifySurfaceFlingerCameraStatus : isFront = false , isOpend = false
I/Camera  (27906): open camera: 1, package name: com.ric.adv_camera_example
I/HwCameraUtil(27906): notifySurfaceFlingerCameraStatus : isFront = true , isOpend = true
I/HwCameraUtil(27906): notifySurfaceFlingerFrontCameraStatus 8011 transact success!
W/System.err(27906): java.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.List.contains(java.lang.Object)' on a null object reference
W/System.err(27906): 	at com.ric.adv_camera.AdvCamera.translateFlashType(AdvCamera.java:507)
W/System.err(27906): 	at com.ric.adv_camera.AdvCamera.setupCamera(AdvCamera.java:469)
W/System.err(27906): 	at com.ric.adv_camera.AdvCamera.onMethodCall(AdvCamera.java:286)
W/System.err(27906): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:230)
W/System.err(27906): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:85)
W/System.err(27906): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:692)
W/System.err(27906): 	at android.os.MessageQueue.nativePollOnce(Native Method)
W/System.err(27906): 	at android.os.MessageQueue.next(MessageQueue.java:363)
W/System.err(27906): 	at android.os.Looper.loop(Looper.java:173)
W/System.err(27906): 	at android.app.ActivityThread.main(ActivityThread.java:8178)
W/System.err(27906): 	at java.lang.reflect.Method.invoke(Native Method)
W/System.err(27906): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
W/System.err(27906): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1101)
`

mcsugaman, 2020-09-12:
I used the version, where front camera was working and then added autofocus implementation...works like a charm

ricnaaru, 2020-12-25:
please use the latest version (1.2.0) and if the problem still persists, reopen this thread. Thanks!

closed 2020-12-25 by ricnaaru