ricnaaru/adv_camera #29: Error opening camera on android emulator

E-Kaese, 2020-06-16:
Code:
```
AdvCamera(
    onCameraCreated: _onCameraCreated,
    onImageCaptured: _onImageCaptured,
)

void _onCameraCreated(AdvCameraController controller) {
    setState(() {
      _cameraController = controller;
    });
}

void _onImageCaptured(String path) {
    print(path);
}
```

Error:
W/System.err(12618): java.lang.NullPointerException: Attempt to invoke interface method 'boolean java.util.List.contains(java.lang.Object)' on a null object reference
W/System.err(12618): 	at com.ric.adv_camera.AdvCamera.translateFlashType(AdvCamera.java:503)
W/System.err(12618): 	at com.ric.adv_camera.AdvCamera.refreshCamera(AdvCamera.java:582)
W/System.err(12618): 	at com.ric.adv_camera.AdvCamera.surfaceChanged(AdvCamera.java:516)
W/System.err(12618): 	at android.view.SurfaceView.updateSurface(SurfaceView.java:770)
W/System.err(12618): 	at android.view.SurfaceView$2.onPreDraw(SurfaceView.java:155)
W/System.err(12618): 	at android.view.ViewTreeObserver.dispatchOnPreDraw(ViewTreeObserver.java:1088)
W/System.err(12618): 	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2745)
W/System.err(12618): 	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1721)

E-Kaese, 2020-06-16:
I presume it has to do with the emulator not having a flash

ricnaaru, 2020-06-21:
hmm did you use the latest version? i though i have fixed this

E-Kaese, 2020-06-21:
Version I'm using: 1.1.1+1

E-Kaese, 2020-07-08:
Any updates on this?

ricnaaru, 2020-08-07:
since this is just on emulator and not really urgent, i may test this in quite some time in the future, will update later

benneca, 2020-08-28:
this also seems to happen on any device that does not have a flash at least the 2 android tablets that I have tried. I have had the same results on a samsung tab a (SM-T510) and a galaxy tab s2 (SM-T713).  I get a black screen when I load the camera. Adding a null check on the supportedFocusMode gets me to a frozen camerapreview but then I get the error listed above. I can modify the translateFlashType function to check if the supported modes are null but I then run into a set parameter failed issue. Would love to see a fix for this as this is on real devices and therefore a pretty high priority.

ricnaaru, 2020-12-25:
thanks @benneca will add it to the code!

closed 2020-12-25 by ricnaaru