ricnaaru/adv_camera #30: No Camera preview

dhanuvanth, 2020-06-27:
no error in the code but it is not showing any preview. I tried with both android emulated and also real device. but its showing only the progress bar. I used the latest version (adv_camera: ^1.1.1+1). android version - 7,8,9
[image]
[image]

please help...

chadathanainta, 2020-07-22:
Add permission in Android

Abhishek01039, 2020-07-27:
can you please let us know which permission we need to give and where? @chadathanainta

chadathanainta, 2020-07-30:
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.FLASHLIGHT" />
    <uses-feature android:name="android.hardware.camera" />
    <uses-feature android:name="android.hardware.camera.front" />
    <uses-feature android:name="android.hardware.camera2" android:required="false" />
    <uses-feature android:name="android.hardware.camera.autofocus" android:required="false"/>
    <uses-feature android:name="android.hardware.camera.flash" android:required="false" />

Abhishek01039, 2020-08-06:
ok thanks @chadathanainta

closed 2020-08-06 by ricnaaru