ricnaaru/adv_camera #34: It's loading forever

panudetjt, 2020-08-13:
```dart
import 'dart:io';

import 'package:adv_camera/adv_camera.dart';
import 'package:flutter/material.dart';

class Camera extends StatefulWidget {
  @override
  State<Camera> createState() {
    return _CameraState();
  }
}

class _CameraState extends State<Camera> {
  List<String> pictureSizes = [];
  String imagePath;

  AdvCameraController cameraController;

  _onCameraCreated(AdvCameraController controller) {
    this.cameraController = controller;

    this.cameraController.getPictureSizes().then((pictureSizes) {
      setState(() {
        this.pictureSizes = pictureSizes;
      });
    });
  }

  @override
  Widget build(BuildContext context) {
    return Stack(
      children: [
        Column(
          children: [
            SingleChildScrollView(
              scrollDirection: Axis.horizontal,
              child: Container(
                color: Colors.purple,
                child: Row(
                  children: [
                    FlatButton(
                      child: Text("Auto"),
                      onPressed: () {
                        cameraController.setFlashType(FlashType.auto);
                      },
                    ),
                    FlatButton(
                      child: Text("On"),
                      onPressed: () {
                        cameraController.setFlashType(FlashType.on);
                      },
                    ),
                    FlatButton(
                      child: Text("Off"),
                      onPressed: () {
                        cameraController.setFlashType(FlashType.off);
                      },
                    ),
                    FlatButton(
                      child: Text("Torch"),
                      onPressed: () {
                        cameraController.setFlashType(FlashType.torch);
                      },
                    ),
                  ],
                ),
              ),
            ),
            SingleChildScrollView(
              scrollDirection: Axis.horizontal,
              child: Container(
                color: Colors.orange,
                child: Row(
                  children: [
                    FlatButton(
                      child: Text(Platform.isAndroid ? "1:1" : "Low"),
                      onPressed: () {
                        cameraController.setPreviewRatio(CameraPreviewRatio.r1);
                        cameraController
                            .setSessionPreset(CameraSessionPreset.low);
                      },
                    ),
                    FlatButton(
                      child: Text(Platform.isAndroid ? "4:3" : "Medium"),
                      onPressed: () {
                        cameraController
                            .setPreviewRatio(CameraPreviewRatio.r4_3);
                        cameraController
                            .setSessionPreset(CameraSessionPreset.medium);
                      },
                    ),
                    FlatButton(
                      child: Text(Platform.isAndroid ? "11:9" : "High"),
                      onPressed: () {
                        cameraController
                            .setPreviewRatio(CameraPreviewRatio.r11_9);
                        cameraController
                            .setSessionPreset(CameraSessionPreset.high);
                      },
                    ),
                    FlatButton(
                      child: Text(Platform.isAndroid ? "16:9" : "Best"),
                      onPressed: () {
                        cameraController
                            .setPreviewRatio(CameraPreviewRatio.r16_9);
                        cameraController
                            .setSessionPreset(CameraSessionPreset.photo);
                      },
                    ),
                  ],
                ),
              ),
            ),
            Container(
              color: Colors.blue,
              child: SingleChildScrollView(
                scrollDirection: Axis.horizontal,
                child: Row(
                  children: this.pictureSizes.map((pictureSize) {
                    return FlatButton(
                      child: Text(pictureSize),
                      onPressed: () {
                        cameraController.setPictureSize(
                            int.tryParse(pictureSize.substring(
                                0, pictureSize.indexOf(":"))),
                            int.tryParse(pictureSize.substring(
                                pictureSize.indexOf(":") + 1,
                                pictureSize.length)));
                      },
                    );
                  }).toList(),
                ),
              ),
            ),
            Expanded(
              child: Container(
                child: AdvCamera(
                  onCameraCreated: _onCameraCreated,
                  onImageCaptured: (String path) {
                    print("onImageCaptured => " + path);
                    if (this.mounted)
                      setState(() {
                        imagePath = path;
                      });
                  },
                  cameraPreviewRatio: CameraPreviewRatio.r16_9,
                ),
              ),
            ),
          ],
        ),
        Positioned(
          bottom: 16.0,
          left: 16.0,
          child: imagePath != null
              ? Container(
                  width: 100.0,
                  height: 100.0,
                  child: Image.file(File(imagePath)))
              : Icon(Icons.image),
        )
      ],
    );
  }
}
```
console
```
W/ActivityThread( 4902): handleWindowVisibility: no activity for token android.os.BinderProxy@4c34d4d
V/ViewRootImpl( 4902): The specified message queue synchronization  barrier token has not been posted or has already been removed
Reloaded 26 of 926 libraries in 1,114ms.
D/DecorView( 4902): onWindowFocusChangedFromViewRoot hasFocus: true, DecorView@2a5f5aa[MainActivit
```

ricnaaru, 2020-12-25:
i notice `Reloaded 26 of 926 libraries in 1,114ms.` does this happen when you hot reload? i need more information about this, what device you're using, and when specifically this error happens, for example when you click something or at the beginning?

mdelem, 2021-01-19:
It might be due to the permissions check failing.

In my case after setting `checkPermissionAtStartup` to `false`, I had to update
    `if (!hasPermission) return Center(child: CircularProgressIndicator());`
to 
    `if (widget.checkPermissionAtStartup && !hasPermission) return Center(child: CircularProgressIndicator());`
 in the build method of `adv_camera.dart`

saaimbiz, 2022-02-05:
I found following permission in the example app, add them to the manifest file.

<uses-permission android:name="android.permission.FLASHLIGHT" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.CAMERA" />

    <uses-feature android:name="android.hardware.camera.front" />
    <uses-feature android:name="android.hardware.camera2" />
    <uses-feature
        android:name="android.hardware.camera.autofocus"
        android:required="true" />
    <uses-feature android:name="android.hardware.camera.flash" />

    <uses-permission android:name="android.permission.READ_PHONE_STATE" />

Remove unwanted ones