ricnaaru/adv_camera #37: Disable storage permission request

Stemo688, 2020-09-24:
The problem I found out with my app is in these lines in 'checkForPermission' method in AdvCameraPlugin.java:
`Dexter.withActivity(activity)
                .withPermissions(Manifest.permission.CAMERA, Manifest.permission.WRITE_EXTERNAL_STORAGE, Manifest.permission.READ_EXTERNAL_STORAGE)`

I suggest that better approach would be this:
`Dexter.withActivity(activity)
                .withPermissions(Manifest.permission.CAMERA)`
Or rather, not requesting storage permission. Here is my reasoning and my personal case:
I implemented a dialog which explain the usage of storage in my app, which is showed before the system dialog of storage permission. The problem is that storage permission for AdvCamera is asked on initialization, so a possible solution is to first show the dialog, then initializate the widget. The problem with this is that storage is not necessary for camera since in a case of permission denial the image could be temporarly saved in a path where storage permission is not required (i.e. Directory.systemTemp), since in my case the image is only used for further processing and upload. But as I checked in the AdvCamera widget, this widget is only initialitied when permissions are successful, which in my case is not good since I would like my users to still take photos even if they don't concede storage permission, if they do the image is just permantly storaged.

So to conclude, for me a better approach would be not to ask storage permissions in both iOS (in this case I didn't check the code) and Android, and leave the duty to ask permissions to the developer with a package like permission_handler (if they need it), or simply just adding an option to disable it in the package.
Thank you and best regards.

ricnaaru, 2020-12-29:
The reason I've added the storage permission is I have the setSavePath function on controller which allow programmer to put custom path for their images. So the solution I can give for this problem is, at the latest version (> 1.3.1), I've added `checkPermissionAtStartup` flag, if you feel you want to override the permission checking process, just pass this flag to false, thus, I will not check the permission at initState. Thanks!

closed 2020-12-29 by ricnaaru

Stemo688, 2021-01-03:
Thanks a lot!

Stemo688, 2021-01-08:
Hi, I cannot re-open this issue, but I found a problem with the implementation. If `checkPermissionAtStartup` is true then `hasPermission` can never be true and so the camera cannot be loaded. Even if `hasPermission` is public it cannot be accessed since the state is private. I have made a pull request (871a2e0) to address this issue, feel free to merge it to master or edit it if necessary.
I tested the changes and work in the latest Flutter version.

ricnaaru, 2021-01-08:
Hi, thanks, i will check it later!