ricnaaru/adv_camera #42: how to set custom focus point?

Eswarrajpurohit, 2020-11-03:
Hey there,
As you have mentioned that there is a tap to focus method through which we can get focus but, as i have a use case where i have to use stack and i want focus on some point for a longer duration but when we tap i can get focus for 1 or 2 sec and the the focus is again removed is there any way by which i can customize the way the focus has been test.
Please help me out

ricnaaru, 2020-12-25:
this is a tough one, especially for android, since both Flutter and Android have different method for pixels, but i'm working on it, let's see

Eswarrajpurohit, 2020-12-26:
Its great that you are working on this awesome project can I also help you
with this?

ricnaaru, 2020-12-26:
sure if you have any fix or idea, just fork it and submit pull request, i will review it later, thanks!

ricnaaru, 2020-12-29:
the manual focus functionality now is available on version 1.3.0. i'll close this thread for now, if there's a problem using that functionality, feel free to reopen this thread.

closed 2020-12-29 by ricnaaru